JeffersonLab/halld_recon #889: Memory leak when analyzing REST files

staylorjlab, 2025-03-10:
Extracting the DBeamPhoton data from a REST file leads to a memory leak that does not occur when running the reconstruction on an evio file.  The output from running valgrind on an hd_dump process that just prints out DBeamPhoton objects gives the following:

==2405964== Memcheck, a memory error detector
==2405964== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==2405964== Using Valgrind-3.23.0 and LibVEX; rerun with -h for copyright info
==2405964== Command: hd_dump -DBeamPhoton -p /cache/halld/RunPeriod-2019-11/recon/ver01/REST/071820/dana_rest_071820_000.hddm -Pjana:warmup_timeout=1000 -Pjana:timeout=1000 -Pjana:nevents=20000
==2405964== Parent PID: 514838
==2405964== 
==2405964== 
==2405964== 172,450,112 (90,765,920 direct, 81,684,192 indirect) bytes in 567,287 blocks are definitely lost in loss record 8,337 of 8,337
==2405964==    at 0x4844FB5: operator new(unsigned long) (vg_replace_malloc.c:487)
==2405964==    by 0x13FC63C: DEventSourceREST::Extract_DBeamPhoton(hddm_r::HDDM*, JFactoryT<DBeamPhoton>*, std::shared_ptr<JEvent const> const&) (DEventSourceREST.cc:741)
==2405964==    by 0x1402381: DEventSourceREST::GetObjects(std::shared_ptr<JEvent const> const&, JFactory*) (DEventSourceREST.cc:365)
==2405964==    by 0x74AC9F8: JFactory::Create(std::shared_ptr<JEvent const> const&)::{lambda()#1}::operator()() const (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964==    by 0x74AD884: void jana::components::JComponent::CallWithJExceptionWrapper<JFactory::Create(std::shared_ptr<JEvent const> const&)::{lambda()#1}>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, JFactory::Create(std::shared_ptr<JEvent const> const&)::{lambda()#1}) (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964==    by 0x74ACD5E: JFactory::Create(std::shared_ptr<JEvent const> const&) (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964==    by 0x7D2A9C: MyProcessor::Process(std::shared_ptr<JEvent const> const&) (MyProcessor.cc:157)
==2405964==    by 0x7D8480: operator() (JEventProcessor.h:142)
==2405964==    by 0x7D8480: void jana::components::JComponent::CallWithJExceptionWrapper<JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&)::{lambda()#3}>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&)::{lambda()#3}) (JComponent.h:142)
==2405964==    by 0x7D89AD: JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&) (JEventProcessor.h:142)
==2405964==    by 0x74CA4AF: JEventMapArrow::fire(JEvent*, std::array<std::pair<JEvent*, int>, 2ul>&, unsigned long&, JArrow::FireResult&) (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964==    by 0x74BEEA8: JExecutionEngine::RunWorker(JExecutionEngine::Worker) (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964==    by 0x74C620A: void std::__invoke_impl<void, void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker>(std::__invoke_memfun_deref, void (JExecutionEngine::*&&)(JExecutionEngine::Worker), JExecutionEngine*&&, JExecutionEngine::Worker&&) (in /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so)
==2405964== 
==2405964== LEAK SUMMARY:
==2405964==    definitely lost: 149,933,080 bytes in 937,059 blocks
==2405964==    indirectly lost: 171,517,437 bytes in 1,875,089 blocks
==2405964==      possibly lost: 2,029,668 bytes in 471 blocks
==2405964==    still reachable: 350,279 bytes in 447 blocks
==2405964==                       of which reachable via heuristic:
==2405964==                         length64           : 176,824 bytes in 412 blocks
==2405964==         suppressed: 23,325,252 bytes in 40,185 blocks
==2405964== Reachable blocks (those to which a pointer was found) are not shown.
==2405964== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==2405964== 
==2405964== Use --track-origins=yes to see where uninitialised values come from
==2405964== For lists of detected and suppressed errors, rerun with: -s
==2405964== ERROR SUMMARY: 69 errors from 49 contexts (suppressed: 1021 from 980)

aaust, 2025-03-11:
I just ran monitoring_hists together with ReactionFilter over a REST file, using the `MONITOR:MEMORY_EVENTS 100000` option.
The results confirm this issue. It is ugly (log scale!)

[image]

The memory footprint grows linearly.

RaiqaRasool, 2025-03-12:
Thanks for confirming this! That'll be helpful for my testing later too when I feel I've solved the issue. I think I've pinpointed a potential cause, but I need to dig deeper to be sure.

aaust, 2025-03-13:
Thank you for this fix. I ran the same test again:
[image]
While the situation is dramatically improved, there still seems to be a smaller memory leak which adds a linear component to the memory footprint.
NB: this plot is no longer in log scale

RaiqaRasool, 2025-03-24:
## **Confirmation of No Memory Leak in JANA2**
### **Previous Behavior (Before Fixing the Memory Leak)**
Before resolving the **NOT_OBJECT_OWNER** memory leak, memory usage **continued increasing over time**, as observed in the `top` output. Virtual memory usage started at **18.0 GB**, then gradually increased to **18.3 GB**, confirming a memory leak.
#### `top`
[image]  
[image]
#### `massif`
This memory leak trend was also evident in the **Massif heap allocation graph**:  
[image]
### **Current Behavior (After Fixing the Memory Leak)**
Now, after resolving the issue, both **JANA1 and JANA2 reach a stable memory usage plateau** instead of continuously increasing, as confirmed by the `top` output.
#### **`top` – Monitoring Hists Only**
_(JANA1 on Top, JANA2 on Bottom)_  
[image]
#### **`top`  – Reaction Filter Only**
_(JANA1 on Top, JANA2 on Bottom)_  
[image]

Observations from the `top` outputs show that both **JANA1 and JANA2** reach a **maximum memory usage of 18.0 GB virtual memory** and then **stop increasing**, confirming that memory allocation remains stable.
#### **`massif`**
A similar trend is observed in the **Massif heap allocation graph** for JANA2:  
[image]
### **Why Does the Total Memory Plot Still Show a Linear Trend?**
Even after fixing the **NOT_OBJECT_OWNER** memory leak, the **total memory plot** generated in the `Hist_ObjectMemory` folder using `-PMONITOR:MEMORY_EVENTS` still shows a **linear trend**, raising concerns about a potential memory leak:

[image]

However, this **does not indicate an actual memory leak**. The reason lies in the **logic used to generate the total memory plot**. This plot does **not** track actual memory allocation but instead sums up the sizes of objects in different dresource pools.
##### **Snippet of Code Responsible for Total Memory Plot Calculation**
_(From `DHistogramAction_ObjectMemory::Perform_Action`)_
```cpp
//DKinematicData::DKinematicInfo
locBin = dBinMap["DKinematicInfo"];
locNumObjectsMap[locBin] = dResourcePool_KinematicInfo.Get_NumObjectsAllThreads();
locMemory = sizeof(DKinematicData::DKinematicInfo) * locNumObjectsMap[locBin];
locMemoryMap[locBin] = locMemory;
locTotalMemory += locMemory;
```

This method does not reflect the program's actual memory usage but simply tracks pool objects count and size.

#### **The Correct Representation of Memory Behavior**
Instead of relying on the **total memory plot**, the **resident memory and virtual memory graphs** provide a more accurate representation of memory behavior.

Previously, due to the **NOT_OBJECT_OWNER memory leak**, these graphs exhibited a **linear increasing trend**. Now, they show the expected behavior: **memory usage rises up to a certain point and then stabilizes**.

**Virtual Memory Graph**
_(Left: Before Fix, Right: After Fix)_  
[image]

**Resident Memory Graph**
_(Left: Before Fix, Right: After Fix)_  
[image]

### **Conclusion**
All observed data—from **`top` output, Massif heap graphs, and memory usage graphs**—confirm that the **memory leak has been resolved**. The memory usage **stabilizes after reaching a peak**, instead of continuously increasing, proving that the program now correctly manages memory allocation.

commit 31d592b2 (2025-03-13): Fix #889 - Memory Leak Due to NOT_OBJECT_OWNER Flag in DBeamPhoton_factory