JeffersonLab/halld_recon #890: CCDB locking on startup in certain cases?

sdobbs, 2025-03-10:
This doesn't show up when running with more plugins, but I noticed that when running several threads with a simple plugin, the command below seemed to lock up on startup (at least 5-10 minutes).  Running on a debugger suggested that the hold up may be somewhere in the CCDB code.

`hd_root -PNTHREADS=8 -PPLUGINS=BEAM_online /cache/halld/RunPeriod-2023-01/rawdata/Run121041/hd_rawdata_121041_033.evio`

RaiqaRasool, 2025-03-26:
Hi @sdobbs,

So after doing some research, what seems to be happening is that when running a simple plugin (whose `::BeginRun` method is mainly responsible for fetching calibration data) with multiple threads, all the `BeginRun` calls for the factory reach the point of querying the CCDB almost simultaneously. However, only one thread can query the CCDB at a time due to the mutex lock:

```cpp
bool GetCalib(string namepath, vector<map<string, string> >& vsvals, uint64_t event_number = 0) {
    // Lock mutex for exclusive use of underlying Calibration object
    pthread_mutex_lock(&mutex);
}
```

Hence, only one thread can actually begin polling for the calibration data after sending the request to the CCDB:
[image]

While all other threads are blocked, waiting for the lock:
[image]

Their backtrace looks like this:  
[image]

So now as fetching calibration constants from the CCDB happens sequentially, adding more threads to the plugin with many CCDB accesses in `::BeginRun` will significantly slow down the startup, as the number of threads increases the number of sequential CCDB calls. 

However, for plugins with minimal CCDB access or those that have other work to do in `::BeginRun` besides just fetching constants, not all threads will be waiting on the CCDB lock. This reduces contention, and as the threads are occupied with other tasks instead of just waiting for the lock, it gives the illusion of the system not getting stuck during startup.

It's important to note that this issue of "getting stuck" in JANA2 is essentially the same as it is in JANA1. Both of them seem to "lock up" when running with a simple plugin and a high number of threads, coupled with a plugin that makes numerous CCDB accesses in `::BeginRun`. To clarify, there isn’t a true "lock-up" happening. Instead, the parallel CCDB access is causing each thread to poll after sending a request to the CCDB, resulting in significant wait times, which can give the appearance that the system isn’t progressing. However, the program is still working; it's just that no events are being processed during the time the threads are waiting for the lock.

I would appreciate insight from @faustus123  and @nathanwbrei  on this as well, to confirm if this is the expected behavior and the potential reason behind the behavior observed by Sean, or if I’m missing something. Additionally, it would be great to explore if we can find a good solution to work around this issue.

sdobbs, 2025-03-26:
Hi Raiqa,  thanks for the detailed explanation!  I guess my only 2 cents is that since I saw the extreme startup delays when asking just for BEAM_online, but not when asking for the combination of BEAM_online and monitoring_hists, it did sound like there was a contention issue.  So it's good to hear confirmation of that!  I guess that what I don't understand is exactly why the delays are so long in the case mentioned in this issue, and get shorter when we add in more plugins.

One question - for the screenshots you showed, how many processing threads did you request?

RaiqaRasool, 2025-03-26:
My guess for the noticeable delays is this:

For plugins with minimal CCDB access or those that have other tasks to perform in `::BeginRun` in addition to just fetching constants, not all threads will be blocked waiting for the CCDB lock. Since these threads are busy with other tasks, such as processing or printing output, it gives the illusion that the system isn’t getting stuck during startup. Even though the same amount of time is ultimately spent, the presence of other threads performing useful work (and potentially updating the screen) makes it seem like the delays aren’t as long.

For testing, I used 32 threads and also counted the number of threads in GDB. The result showed 32 worker threads, plus 4 EVIO threads, and I believe there is one main thread, making the total thread count 37.

RaiqaRasool, 2025-03-28:
Hi @sdobbs,

I’ve noticed that the delay effect persists even with `monitoring_hists` when it is being run with other plugins that have factories attached, which are fetching data from CCDB quite frequently. So this behavior is consistent. As long as, we are using the factories with multiple ccdb calls inside the `::BeginRun` this delay will be observed.

RaiqaRasool, 2025-03-31:
I think, I’d like to keep this issue open, as improving how we access CCDB could potentially enhance the rate and may also be the reason behind the rate drop when increasing the number of threads beyond a certain point.  

@nathanwbrei, would you prefer to close this issue and create a new one, or attach the relevant commit to this issue once we optimize the access? I'm fine with either approach.

RaiqaRasool, 2025-04-14:
Hi @sdobbs,

I tried removing the locks from all the GetCalib function definitions in JANA2/src/libraries/JANA/Calibrations/JCalibrationCCDB.h, and still observed that getting stuck for 9 to 10 minutes. This time, the threads were getting stuck waiting on the lock in the CCDB::GetAssignment function. 

## Getting Calib Constants
The calibration constants are fetched by calling JANA's GetCalib (`JANA2/src/libraries/JANA/Calibrations/JCalibrationCCDB.h`)
```cpp
    bool GetCalib(string namepath, vector<map<string, string> >& vsvals, uint64_t event_number = 0) {
        try {
            bool result = mCalibration->GetCalib(vsvals, namepath);
            return !result; 
        }
        catch (std::exception& ex) {
            return true;
        }
    }

```

JANA GetCalib calls CCDB GetCalib (`ccdb/cpp/src/CCDB/Calibration.cc`)
```cpp
bool Calibration::GetCalib( vector< map<string, string> > &values, const string & namepath )
{
   auto assignment = GetAssignment(namepath, true);
        
    if(!assignment)
    {       
        return false; 
    }
    assert(values.empty());
    assignment->GetMappedData(values);
    
    if(values.size() == 0){
        throw std::logic_error("Calibration::GetCalib( vector< map<string, string> >&, const string&). Data has no rows. Zero rows are not supposed to be.");
    }
    return true;
}
```
CCDB GetCalib  calls CCDB GetAssignment (`ccdb/cpp/src/CCDB/Calibration.cc`) which locks a mReadMutex lock:

```cpp
Assignment* Calibration::GetAssignment(const string& namepath, bool loadColumns /*=true*/)
{
    // GetAssignment code .....

    std::lock_guard<std::mutex> lock(mReadMutex);

    // Check if we have this value in the cache
    string cache_key = namepath + ":" + to_string(run) + ":" + variation + ":" + to_string(time);
    std::cout<<"Calibration -- mIsCacheEnabled: "<<mIsCacheEnabled<<std::endl;
    if(mIsCacheEnabled)
    {
        if (  cache.find(cache_key) != cache.end() ) {
            return cache[cache_key];
        }
    }

    Assignment* assigment;

    if(time < 0) time = 0;

    assigment = (mProvider->GetAssignmentShort(run, PathUtils::MakeAbsolute(result.Path), time, variation, loadColumns));

    if(mIsCacheEnabled) {
        cache[cache_key] = assigment;
    }

    return assigment;
}
```
So even though the lock was removed from JANA, threads still cannot execute in parallel due to the lock in the CCDB API. This indicates that the issue is not related to JANA’s locking mechanism, as the rates remain the same with or without the JANA locks.

## CCDB Cache Enabling
It's important to note that after acquiring the lock, the function (`GetAssignment`) either looks for data in the cache or fetches it directly from the database, depending on whether caching is enabled. Enabling caching has had a significant impact on performance. Even though calibration constants are still retrieved sequentially due to the lock being held for each fetch, the cache effectively eliminates the need to repeatedly fetch data from the database and instead get it locally from cache. This has drastically reduced the overhead associated with large startup times and resulted in a substantial increase in performance. Here are the test results that highlight the improvements:

<p>For 1000 events and 32 threads, with <code inline="">monitoring_hists</code> and <code inline="">BEAM_online</code> plugins:</p>

<h5>Event Processing Rates:</h5>

Configuration | Cache ON | Cache OFF
-- | -- | --
monitoring_hists + BEAM_online | 19 Hz | 1.2 Hz
BEAM_online only | 74.7 Hz | 1.7 Hz

<h5>Startup Time:</h5>

|Configuration | Cache ON | Cache OFF
| ------------------------------ | --- | ------ |
| BEAM_online only               | 11s | 10 min |

### Memory Usage
#### Cache OFF (left) versus ON (right):</p>
[image]

#### 1k vs. 100k Events (Cache ON)
Given following remarks on `EnableCache` function of the `Calibration` class by CCDB author:

> cache greatly (2 magnitudes) reduces the time to get the same constants from DB
    but it costs some memory. Shouldn't be a bug source but caches are always caches

To confirm that enabling caching is still keeping memory usage within acceptable limits, we compared memory usage for 1k events versus 100k events. The results show that memory usage increased by only 500MB with a 100x increase in events, suggesting that the memory cost is manageable. Given the substantial performance gains provided by caching, this slight memory increase is well within acceptable limits.

[image]

<h3>Conclusion:</h3>
<p>Enabling <strong>CCDB caching</strong> offers following benefits:</p>
<ul>
<li>
<p><strong>Significant rate improvements</strong>: Substantial boost in event processing rates.</p>
</li>
<li>
<p><strong>Faster startup times</strong>: Reduced startup time due to fewer database accesses.</p>
</li>
<li>
<p><strong>Reduced memory usage</strong>: Memory usage remains efficient even with larger event sets.</p>
</li>
</ul>
<p>As a result, CCDB caching will be enabled by default in the next JANA2 release, and users will experience these benefits automatically.</p>

sdobbs, 2025-04-14:
Thanks so much for these studies, @RaiqaRasool !  Enabling CCDB caching sounds like a great idea.

Do we want to keep this issue open for any reason (like any checks on our side?)

aaust, 2025-04-18:
I built [version_6.1.3.xml](https://halldweb.jlab.org/halld_versions/version_6.1.3.xml) with [jana2.4.1](https://github.com/JeffersonLab/JANA2/releases/tag/v2.4.1). All other packages are exactly the same as in [version_6.1.2.xml](https://halldweb.jlab.org/halld_versions/version_6.1.2.xml). When running monitoring_hists on the interactive farm with 32 thread over 1000 events, I can confirm this impressive improvement:
|  | jana2.4.0  | jana2.4.1 |
| - | ------------- | ------------- |
| real time | 14m44 | 0m40s  |
| average rate | 1.1Hz  | 26Hz  |

The average rate includes setting up all threads. This improvement is washed out when using the full suite of monitoring plugins and running over more events, but for example with 10k events, it is still considerable:
[image]

The difference gets even smaller with 100k events. For low number of threads, the caching during startup is probably outweighed by other factors during the >4h long process that cannot be controlled:
[image]

The region between 32 and 64 threads still seems to be optimal. @RaiqaRasool proposed to vary the number of evio threads for these jobs, but this goes beyond the scope of this issue. I propose to close this issue, as a large improvement has been achieved.

sdobbs, 2025-04-18:
this looks great, I agree we should close this issue.

If we want to make progress on the launches with a full complement of plugins, we should also investigate the impact of disk I/O (skims and trees) on performance

closed 2025-04-18 by sdobbs