JeffersonLab/halld_recon #892: crash with ECAL data

aaust, 2025-03-10:
Many plugins crash with recent raw data from 2025. The row/column assignment looks bad:

```
17:07:30.536 [fatal] Exception in worker thread 0 while executing PhysicsEventMap2 on event #17662161
----------------------------------------------------------
JException
  Message:  DECALHit: Loading ECAL constant for inactive channel!  row=65537, col=113
  Function: JFactory::BeginRun
  Class:    JEventProcessor_FCAL_online
  Instance: DECALHit
  Plugin:   FCAL_online
  Backtrace:

   0: ???
      hd_root:311444

   1: DECALHit_factory::BeginRun(std::shared_ptr<JEvent const> const&)
      hd_root:b4b5cb
```

I have seen it with many highlevel plugins: highlevel_online, FCAL_online, CDC_dedx

How to reproduce it:
`hd_root -PNTHREADS=8 -PPLUGINS=FCAL_online /cache/halld/RunPeriod-2025-01/rawdata/Run130437/hd_rawdata_130437_017.evio`

zihlmann, 2025-03-11:
confirm: but running single thread with gdb no stack???
...
Closed ROOT file
Exit code: 1
[Inferior 1 (process 1211701) exited with code 01]
Missing separate debuginfos, use: dnf debuginfo-install bzip2-libs-1.0.8-10.el9_5.x86_64 expat-2.5.0-3.el9_5.1.x86_64 flexiblas-netlib-3.0.4-8.el9.x86_64 flexiblas-openblas-openmp-3.0.4-8.el9.x86_64 freetype-2.10.4-9.el9.x86_64 glib2-2.68.4-14.el9_4.1.x86_64 glibc-2.34-125.el9_5.1.alma.2.x86_64 graphite2-1.3.14-9.el9.x86_64 harfbuzz-2.7.4-10.el9.x86_64 keyutils-libs-1.6.3-1.el9.x86_64 krb5-libs-1.21.1-4.el9_5.x86_64 libbrotli-1.0.9-7.el9_5.x86_64 libcom_err-1.46.5-5.el9.x86_64 libgcc-11.5.0-5.el9_5.alma.1.x86_64 libgfortran-11.5.0-5.el9_5.alma.1.x86_64 libgomp-11.5.0-5.el9_5.alma.1.x86_64 libicu-67.1-9.el9.x86_64 libnsl2-2.0.0-1.el9.x86_64 libpng-1.6.37-12.el9.x86_64 libquadmath-11.5.0-5.el9_5.alma.1.x86_64 libselinux-3.6-1.el9.x86_64 libstdc++-11.5.0-5.el9_5.alma.1.x86_64 libtirpc-1.3.3-9.el9.x86_64 libzstd-1.5.1-2.el9.x86_64 lz4-libs-1.9.3-5.el9.x86_64 mariadb-connector-c-3.2.6-1.el9_0.x86_64 openblas-openmp-0.3.26-2.el9.x86_64 openssl-libs-3.2.2-6.el9_5.1.x86_64 pcre-8.44-4.el9.x86_64 pcre2-10.40-6.el9.x86_64 sssd-client-2.9.5-4.el9_5.4.x86_64 xxhash-libs-0.8.2-1.el9.x86_64 xz-libs-5.2.5-8.el9_0.x86_64 zlib-1.2.11-40.el9.x86_64
(gdb) bt
No stack.
(gdb)

zihlmann, 2025-03-11:
row=65537: indicative of uninitialized variable

zihlmann, 2025-03-11:
Just realize now: Why is there a mix of FCAL and ECAL notation? Plugin FCAL_online but Instance DECALHit? Wouldn't such a choice lead to confusion?

staylorjlab, 2025-03-11:
This plugin asks for the FCAL showers, which need the ECAL hits if they are present.

zihlmann, 2025-03-11:
That makes sense, thanks! but the problem seems to happen/start in DECALHit_factory ... or not?

From Alex:
   1: DECALHit_factory::BeginRun(std::shared_ptr<JEvent const> const&)
      hd_root:b4b5cb

I see this (and more):
   0: ???
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/halld_recon/halld_recon-5.0.0/Linux_Alma9-x86_64-gcc11.5.0/bin/hd_root:311444

   1: **DECALHit_factory**::BeginRun(std::shared_ptr<JEvent const> const&)
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/halld_recon/halld_recon-5.0.0/Linux_Alma9-x86_64-gcc11.5.0/bin/hd_root:b4b5cb

   2: ???
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1f5a9b

   3: ???
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1f7921

   4: JFactory::Create(std::shared_ptr<JEvent const> const&)
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1f5f6c

   5: JFactoryT<DECALHit>* JEvent::Get<DECALHit>(std::vector<DECALHit const*, std::allocator<DECALHit const*> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) const
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/halld_recon/halld_recon-5.0.0/Linux_Alma9-x86_64-gcc11.5.0/bin/hd_root:b281b6

   6: DFCALCluster_factory::Process(std::shared_ptr<JEvent const> const&)
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/halld_recon/halld_recon-5.0.0/Linux_Alma9-x86_64-gcc11.5.0/bin/hd_root:b2678e

   7: ???
      /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1f5ad3

zihlmann, 2025-03-11:
Simon works on a fix. There are some inconsistencies in the code that need to be modified or removed.

aaust, 2025-03-11:
Fixed by #893 , thank you @staylorjlab

closed 2025-03-11 by aaust