JeffersonLab/halld_recon #898: Using Custom Plugin Over .evio files from evio_writer fails

alfab3, 2025-03-13:
I made custom evio files using this custom plugin:
/w/halld-scshelf2101/home/alfab/cpp_analysis/jana2_plugins/track2skim/ (which calls on: src/plugins/Utilities/evio_writer/DEventWriterEVIO.cc)

When running a custom plugin over these newly made files outputted from the evio_writer the Event Processor fails:

13:17:18.918 [fatal] Segfault detected!
13:17:18.918 [fatal] Hard exit due to segmentation fault! Backtrace:
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    0: DEVIOWorkerThread::Parsef125Bank(unsigned int, unsigned int*&, unsigned int*)
13:17:18.918 [fatal]       hd_root:e2cd4f
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    1: DEVIOWorkerThread::ParseJLabModuleData(unsigned int, unsigned int*&, unsigned int*)
13:17:18.918 [fatal]       hd_root:e2fe51
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    2: DEVIOWorkerThread::ParseDataBank(unsigned int*&, unsigned int*)
13:17:18.918 [fatal]       hd_root:e34918
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    3: DEVIOWorkerThread::ParsePhysicsBank(unsigned int*&, unsigned int*)
13:17:18.918 [fatal]       hd_root:e35077
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    4: DEVIOWorkerThread::ParseBank()
13:17:18.918 [fatal]       hd_root:e367d0
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    5: DEVIOWorkerThread::MakeEvents()
13:17:18.918 [fatal]       hd_root:e375f3
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    6: DEVIOWorkerThread::Run()
13:17:18.918 [fatal]       hd_root:e3aafc
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    7: ???
13:17:18.918 [fatal]       /lib64/libstdc++.so.6:dbad4
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    8: ???
13:17:18.918 [fatal]       /lib64/libc.so.6:89d22
13:17:18.918 [fatal] 
13:17:18.918 [fatal]    9: ???
13:17:18.918 [fatal]       /lib64/libc.so.6:10ed40

However the same custom plugin works over RAW data.

alfab3, 2025-03-13:
UPDATE: 
Testing with a blank plugin:
/w/halld-scshelf2101/home/alfab/cpp_analysis/jana2_plugins/tester/
All skim files produced from evio_writer seem to exhibit this behavior (same error as above):
hd_root -Pplugins=tester /volatile/halld/home/alfab/RunPeriod2022-05/twotrackskims_jana2/twotracks/101142/ twotracks_101142_000.evio - FAILED
hd_root -Pplugins=tester /cache/halld/RunPeriod-2022-05/recon/ver01/cpp_2c_skim/100531/cpp_2c_100531_000.evio - FAILED
hd_root -Pplugins=tester /cache/halld/RunPeriod-2022-05/skims/empty/101167_101254/pipi/skim_101232.pipi.evio - FAILED

aaust, 2025-03-13:
Thank you for reporting this, I was able to confirm the issue with the current master. It does not crash with version_6.0.0.xml. When I revert PR #880 in the master branch, it works again. @nsjarvis , are your changes intended to be backwards-compatible?

nsjarvis, 2025-03-13:
Yes, of course.

aaust, 2025-03-13:
Something must be different in the evio skims than in the original evio files:
```
17:32:26.409 [error]  Truncated f125 FDC hit (missing continuation word!)
17:32:26.409 [error]  Truncated f125 FDC hit (missing continuation word!)
17:32:26.409 [error]  Truncated f125 FDC hit (missing continuation word!)
17:32:26.409 [fatal] Segfault detected!
17:32:26.409 [fatal] Hard exit due to segmentation fault! Backtrace:
17:32:26.409 [fatal] 
17:32:26.409 [fatal]    0: DEVIOWorkerThread::Parsef125Bank(unsigned int, unsigned int*&, unsigned int*)
17:32:26.409 [fatal]       hd_root:e2cd5f
17:32:26.409 [fatal] 
17:32:26.409 [fatal]    1: DEVIOWorkerThread::ParseJLabModuleData(unsigned int, unsigned int*&, unsigned int*)
17:32:26.409 [fatal]       hd_root:e2fe61
17:32:26.409 [fatal] 
17:32:26.409 [fatal]    2: DEVIOWorkerThread::ParseDataBank(unsigned int*&, unsigned int*)
17:32:26.409 [fatal]       hd_root:e34928
```

sdobbs, 2025-03-13:
Probably, the output routines should be updated and were doing something
not entirely correct previously.

nsjarvis, 2025-03-14:
This is a previously unnoticed hardware fault that is not handled properly.   Thank you @alfab3  for spotting this and reporting it!  

I found it in event 4488 of /cache/halld/RunPeriod-2022-05/skims/empty/101167_101254/pipi/skim_101232.pipi.evio
This is what I see with 1 evio and 1 jana thread and verbose output:
```
    FADC125 Block Header: slot=9
      FADC125 Event Header: itrigger=52929 last_itrigger=4294967294, rocid=52, slot=9
      FADC125 Trigger Time (t=1778385089)
      FADC125 FDC Pulse Data(peak) word1: cbe49910
      FADC125 FDC Pulse Data (chan=62 pulse=9 time=401 QF=0 OC=0)
      FADC125 FDC Pulse Data(peak) word2: ca1706f
      FADC125 FDC Pulse Data (integral=0 time=46 pedestal=111)
20:47:25.119 [error]  Truncated f125 FDC hit (missing continuation word!)
      FADC125 FDC Pulse Data(peak) word2: 21b17068
      FADC125 FDC Pulse Data (integral=0 time=46 pedestal=104)
20:47:25.119 [error]  Truncated f125 FDC hit (missing continuation word!)
      FADC125 FDC Pulse Data(peak) word2: 13f17061
      FADC125 FDC Pulse Data (integral=0 time=46 pedestal=97)
20:47:25.119 [error]  Truncated f125 FDC hit (missing continuation word!)
      FADC125 FDC Pulse Data(peak) word2: 5d96864
      FADC125 FDC Pulse Data (integral=0 time=45 pedestal=100)
20:47:25.119 [error]  Truncated f125 FDC hit (missing continuation word!)
20:47:25.119 [error]  Truncated f125 FDC hit (missing continuation word!)
      FADC125 Event Header: itrigger=52929 last_itrigger=4294967294, rocid=52, slot=9
      FADC125 Trigger Time (t=1778385089)
20:47:25.119 [fatal] Segfault detected!
20:47:25.119 [fatal] Hard exit due to segmentation fault! Backtrace:
20:47:25.119 [fatal] 
20:47:25.119 [fatal]    0: DEVIOWorkerThread::Parsef125Bank(unsigned int, unsigned int*&, unsigned int*)
20:47:25.119 [fatal]       hd_root:e34c1f
20:47:25.119 [fatal] 
etc
```

The FDC output should be in pairs of words, and the problem can be seen in the translation of the first word
```
      FADC125 FDC Pulse Data(peak) word1: cbe49910
      FADC125 FDC Pulse Data (chan=62 pulse=9 time=401 QF=0 OC=0)
```
which claims that it has data for NPK=9 (pulse=9) peaks, but I know that was not true as the FDC firmware has ALWAYS been configured to output data for only 1 peak per hit.   Before my recent PR, to read in the extra peaks that the TRD firmware is presently configured for, the code assumed that this number NPK was always 1 and so this hardware fault went unnoticed.  I can see how to modify the code to cope with this better.

nsjarvis, 2025-03-14:
Please try the branch nsj_fix_fa125readout.  I made a [PR](https://github.com/JeffersonLab/halld_recon/pull/900) for it.

nsjarvis, 2025-03-14:
I found this when I ran through the rest of the same skim file with the fixed code (I think previously the third one of these caused the crash) :
```
23:30:18.703 [error] Truncated f125 FDC hit (missing continuation word) from rocid=62 slot=7 chan=66 pulse_number=9
23:30:29.031 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=7 chan=54 pulse_number=9
23:30:35.268 [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=9 chan=62 pulse_number=9
23:30:51.857 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=17 chan=48 pulse_number=25
23:31:00.239 [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=18 chan=56 pulse_number=9
23:31:00.968 [error] Truncated f125 FDC hit (missing continuation word) from rocid=56 slot=3 chan=24 pulse_number=9
23:31:03.115 [error] Truncated f125 FDC hit (missing continuation word) from rocid=55 slot=7 chan=62 pulse_number=9
23:31:06.973 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=3 chan=14 pulse_number=17
23:31:13.970 [error] Truncated f125 FDC hit (missing continuation word) from rocid=61 slot=19 chan=54 pulse_number=9
23:31:16.924 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=16 chan=30 pulse_number=9
23:31:26.400 [error] Truncated f125 FDC hit (missing continuation word) from rocid=59 slot=16 chan=62 pulse_number=17
23:31:27.297 [error] Truncated f125 FDC hit (missing continuation word) from rocid=55 slot=9 chan=60 pulse_number=9
23:31:32.154 [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=18 chan=52 pulse_number=9
23:31:38.231 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=3 chan=28 pulse_number=9
23:31:40.490 [error] Truncated f125 FDC hit (missing continuation word) from rocid=62 slot=10 chan=60 pulse_number=9
23:31:43.324 [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=18 chan=66 pulse_number=9
23:31:44.661 [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=8 chan=48 pulse_number=9
23:31:48.993 [error] Truncated f125 FDC hit (missing continuation word) from rocid=53 slot=10 chan=66 pulse_number=9
23:32:22.226 [error] Truncated f125 FDC hit (missing continuation word) from rocid=58 slot=8 chan=68 pulse_number=25
23:32:31.918 [error] Truncated f125 FDC hit (missing continuation word) from rocid=59 slot=10 chan=67 pulse_number=17
```

aaust, 2025-03-14:
Great, it works for me. Were the truncated words in the skim already present in the original evio file?

nsjarvis, 2025-03-14:
There aren't any genuinely truncated hits in the skim.   I did not look in the original evio file, I'm sure it's the same there. 

The fadc accidentally reported a large number of pulses when it only had one pulse to report.  I'm saying accidentally because the fadc was told to find just one pulse per hit, and somewhere that number was corrupted (gained a bit) on the way out of the fadc.  It's either a firmware or a hardware error.  Prior to the TRD readout, we were not using the pulse_number at all, as we knew that it was always = 1.

aaust, 2025-03-14:
@nsjarvis Thanks for the fix and the explanation

@alfab3 Could you please merge the pull request #900 if it solved your problem?

closed 2025-03-17 by alfab3