JeffersonLab/halld_recon #911: More problems with CCDB on startup

sdobbs, 2025-03-24:
When running some batch jobs, I noticed that sometimes the jobs seem to fail when loading info from the CCDB.

It's not clear to me what is going on, but the memory allocation seems to skyrocket, and sometimes I get stack traces showing a large number of threads have been created (often 50 or more - the most I've seen is around 500).

I've put the script used for these jobs and some sample log files in this directory: 
`/volatile/halld/home/sdobbs/data_skim_ps/bad_job_logs`

Please note that the error:
`17:26:26.307 [error] Cannot load Start Counter geometry information!`
is particular to this run period and can be ignored.

sdobbs, 2025-03-24:
I should also mention that this seems to happen around 20% of the time for these jobs

RaiqaRasool, 2025-03-31:
Hi @sdobbs,

I have fixed this issue in #917. Can you please have a look at that and close this issue if you are satisfied with the solution?

Thanks

commit 2173aa08 (2025-03-31): Merge pull request #917 from JeffersonLab/rasool_fixProgramCrashing

Fix #911 - Program crashing with multiple evio input files
commit f3b73e46 (2025-03-31): Fix #911 - Program crashing with multiple evio input files
closed 2025-03-31 by sdobbs