JeffersonLab/halld_recon #919: Segfault in DEventWriterROOT when loading DBeamHelicity objects

sdobbs, 2025-03-31:
The analysis ROOT tree creation seems to crash now that we have added electron beam helicity information to the output format.
I'm creating this issue as documentation while I look into it more, but additional insight would be appreciated!  As far as I can tell, the crash seems to be happening somewhere where factories are being handled when retrieving DBeamHelicity objects, and strangely the HELI_online plugin, which also tries to fetch DBeamHelicity objects, doesn't seem to cause this crash.

The crash seems to happen at line 1355 of src/libraries/ANALYSIS/DEventWriterROOT.cc in the (recently updated) branch sdobbs_helicity_jana2 and from my investigation so far, the crash seems to happen before the DBeamHelicity factory gets called.

`hd_root -PNTHREADS=8 -PPLUGINS=ReactionFilter -PReaction1=1_14__14_8_9 /cache/halld/RunPeriod-2019-11/recon/ver01/REST/071836/dana_rest_071836_100.hddm`

Stack trace:

<pre>
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    0: DEventWriterROOT::Fill_DataTree(std::shared_ptr<JEvent const> const&, DAnalysis::DReaction const*, std::deque<DAnalysis::DParticleCombo const*, std::allocator<DAnalysis::DParticleCombo const*> >&) const
10:13:57.878 [fatal]       hd_root:4adf87
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    1: DEventWriterROOT::Fill_DataTrees(std::shared_ptr<JEvent const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const
10:13:57.878 [fatal]       hd_root:4b0832
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    2: DEventProcessor_ReactionFilter::Process(std::shared_ptr<JEvent const> const&)
10:13:57.878 [fatal]       /work/halld2/home/sdobbs/Software/halld_recon/halld_recon.jana2.git/Linux_Alma9-x86_64-gcc11.5.0/plugins/ReactionFilter.so:52c47
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    3: void jana::components::JComponent::CallWithJExceptionWrapper<JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&)::{lambda()#3}>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&)::{lambda()#3})
10:13:57.878 [fatal]       hd_root:3c5451
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    4: JEventProcessor::DoLegacyProcess(std::shared_ptr<JEvent const> const&)
10:13:57.878 [fatal]       hd_root:3c597e
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    5: JEventMapArrow::fire(JEvent*, std::array<std::pair<JEvent*, int>, 2ul>&, unsigned long&, JArrow::FireResult&)
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:2134b0
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    6: JExecutionEngine::RunWorker(JExecutionEngine::Worker)
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:207ea9
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    7: void std::__invoke_impl<void, void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker>(std::__invoke_memfun_deref, void (JExecutionEngine::*&&)(JExecutionEngine::Worker), JExecutionEngine*&&, JExecutionEngine::Worker&&)
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:20f20b
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    8: std::__invoke_result<void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker>::type std::__invoke<void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker>(void (JExecutionEngine::*&&)(JExecutionEngine::Worker), JExecutionEngine*&&, JExecutionEngine::Worker&&)
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:20f12c
10:13:57.878 [fatal] 
10:13:57.878 [fatal]    9: void std::thread::_Invoker<std::tuple<void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker> >::_M_invoke<0ul, 1ul, 2ul>(std::_Index_tuple<0ul, 1ul, 2ul>)
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:20f05f
10:13:57.878 [fatal] 
10:13:57.878 [fatal]   10: std::thread::_Invoker<std::tuple<void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker> >::operator()()
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:20effc
10:13:57.878 [fatal] 
10:13:57.878 [fatal]   11: std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (JExecutionEngine::*)(JExecutionEngine::Worker), JExecutionEngine*, JExecutionEngine::Worker> > >::_M_run()
10:13:57.878 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:20efe0
10:13:57.878 [fatal] 
10:13:57.878 [fatal]   12: ???
10:13:57.878 [fatal]       /lib64/libstdc++.so.6:dbad4
10:13:57.878 [fatal] 
10:13:57.878 [fatal]   13: ???
10:13:57.878 [fatal]       /lib64/libc.so.6:897e2
10:13:57.878 [fatal] 
10:13:57.878 [fatal]   14: ???
10:13:57.878 [fatal]       /lib64/libc.so.6:10e800
10:13:57.878 [fatal] 
10:13:57.878 [fatal] 
</pre>

sdobbs, 2025-03-31:
@RaiqaRasool - Just wanted to bring this to your attention, if you have some time to look into this

RaiqaRasool, 2025-03-31:
Sounds good!

sdobbs, 2025-03-31:
So, the underlying problem is that I forgot that `Get()/GetSingle()` are now templated functions, so this function call works:

`auto locBeamHelicity = locEvent->GetSingle<DBeamHelicity>("CORRECTED");`

whereas this does not:
<pre>
const DBeamHelicity* locBeamHelicity = NULL;
locEvent->GetSingle(locBeamHelicities, "CORRECTED");
</pre>
(crash also happened for non-tagged objects)

even though this code was in the previous line:
<pre>
const DTrigger* locTrigger = NULL;
locEvent->GetSingle(locTrigger);
</pre>
I'm guessing that's a compatibility thing that we will have to do away with at some point.

sdobbs, 2025-03-31:
I should mention for completeness that asking for a `vector<const DBeamHelicity *>` with `Get()` worked okay.

Just adding notes so that future me / other devs may come across this if a similar problem occurs.

PR #921 (2025-03-31): Fix wrong use of GetSingle() that was causing crashes in DEventWriterROOT
also load a tagged factory, which got accidentally left out of the previous PR

This resolves #919
closed 2025-03-31 by aaust