JeffersonLab/halld_recon #920: Slot from SSP/DIRC event header does not match slot from last block header

aaust, 2025-03-31:
The latest monitoring launch 2023-01 ver13 with version_6.1.0.xml showed failures for the files:

- 120578_000
- 120697_000
- 121119_000

The error messages from the parsing all look similar:

```
09:50:41.071  [warn] Status: 1283 events processed at 16.0 Hz (3.6 Hz avg)
09:50:41.103 [error] Slot from SSP/DIRC event header does not match slot from last block header (20 != 5)
09:50:41.103 [fatal] Segfault detected!
09:50:41.103 [fatal] Hard exit due to segmentation fault! Backtrace:
09:50:41.103 [fatal] 
09:50:41.103 [fatal]    0: DEVIOWorkerThread::ParseSSPBank(unsigned int, unsigned int*&, unsigned int*)
09:50:41.103 [fatal]       hd_root:ef9eb9
09:50:41.103 [fatal] 
09:50:41.103 [fatal]    1: DEVIOWorkerThread::ParseDataBank(unsigned int*&, unsigned int*)
09:50:41.103 [fatal]       hd_root:eff8c8
09:50:41.103 [fatal] 
09:50:41.103 [fatal]    2: DEVIOWorkerThread::ParsePhysicsBank(unsigned int*&, unsigned int*)
09:50:41.103 [fatal]       hd_root:effe77
```

jrstevenjlab, 2025-03-31:
Huh, I don't see any issues noted in the logbook for these runs.  Is this a new problem with JANA2, or was it there in previous monitoring launches?  Also, is it just the first file in the run or all files in the run that fail?

aaust, 2025-03-31:
The other files in the monitoring launch, e.g. 120578_0{24,48,72,96} were all processed without errors.

The error appears also with jana1 (version_5.22.1.xml):
```
JANA ERROR>>Slot from SSP/DIRC event header does not match slot from last block header (20 != 5)
JANA ERROR>> didn't sleep full 0.5 seconds!ead)  12.0Hz  (avg.: 10.7Hz)     
  1317.0 events processed  (1327.0 events read)  12.0Hz  (avg.: 10.7Hz)     
===========================================================
#5  0x00000000012be947 in DEVIOWorkerThread::ParseSSPBank(unsigned int, unsigned int*&, unsigned int*) ()
#6  0x00000000012c298b in DEVIOWorkerThread::ParseDataBank(unsigned int*&, unsigned int*) ()
#7  0x00000000012c2f47 in DEVIOWorkerThread::ParsePhysicsBank(unsigned int*&, unsigned int*) ()
#8  0x00000000012c4180 in DEVIOWorkerThread::ParseBank() ()
```

jrstevenjlab, 2025-04-02:
@sdobbs pointed out that we could implement a similar flag to `SKIP_F250_FORMAT_ERROR` for the SSP to skip over these events rather than having a hard failure for the whole file.  I'll look into adding this

jrstevenjlab, 2025-04-03:
PR https://github.com/JeffersonLab/halld_recon/pull/923 addresses this issue with a new command line flag `EVIO:SKIP_SSP_FORMAT_ERROR` that I tested with file 120697_000.  There appears to only be one error in that file and the parsing continues without issue after that.

Separately, I did notice another error that seemed to be persistent throughout the file which were errors like this one:

` [error] Truncated f125 FDC hit (missing continuation word) from rocid=52 slot=8 chan=28 pulse_number=9`

Apparently they these errors are skipped by default because it didn't cause the parsing to fail.   Is this a known issue that should be ignored or is a persistent issue of this kind something that needs to be investigated?

sdobbs, 2025-04-03:
Thanks, Justin.

Indeed those f125 problems are known and understood - basically there's bad
board that will incorrectly sometimes say there are additional pulses
(that's the pulse_number=9), but it's configured to only write one out.

aaust, 2025-04-04:
I checked the files 001-004 for these 3 runs but the error did not appear again. The faulty files can be processed successfully with Justin's patch.