JeffersonLab/halld_recon #922: Reaction filter doesn't write deuteron thrown information to the output tree

frankboyu, 2025-04-02:
I am looking at the coherent phi photoproduction on deuteron with the deuterium target data from the 2021-11 (SRC) dataset. I used a generator to generate the events via the K+K- decay channel and then the MCWrapper to simulate the detector response. Then in the reaction filter step, which used the following reaction to skim the events

Reaction1                       1_45__11_12_45
Reaction1:Flags             B4_F4

the thrown information in the output PART file only has the two thrown kaons, not the thrown deuteron.

You can find the relevant files at /work/halld2/home/boyu/src_analysis/sim/output/phi_d_2H_test
Specifically, in hddm/dana_rest_gen_coherent_090213_000.hddm, which is the REST file after reconstruction, the deuteron thrown information is still there.

[image]

Then in root/trees/tree_gd_kpkmd__B4_F4_gen_coherent_090213_000.root, you can see that NumThrown is 2 for all the events.

[image]

And these two particles are two kaons indeed, as I checked their invariant mass.

[image]

Below are what I tried to solve this but no success:
1. Try to add the deuteron mass hypothesis (45) to the dana_rest plugin for reconstruction, as we didn't add that in the first full reconstruction of data. The problem is still there.
2. Add the deuteron option in the [mc_tree plugin](https://github.com/JeffersonLab/halld_recon/tree/master/src/plugins/Analysis/mc_tree), which I thought might be related to the output of thrown information. But it didn't solve the problem.

As I glanced through the halld_recon code, it seems the support for deuteron detection is limited, and this thrown information issue may only be one of them. I do hope that I can help improve this in any way I can. Any suggestions or comments are much appreciated.

Bo

aaust, 2025-04-15:
This seems like a bug. The PART trees are created by the ANALYSIS library. Please have a look at [DEventWriterRoot.cc](https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/ANALYSIS/DEventWriterROOT.cc)

aaust, 2025-04-15:
Oh yes, that makes sense. The multiplexing is precisely used to encode the thrown particle id. You should try to include the Deuteron in this function, and of course also in the corresponding DemultiplexPID such that the DSelector can interpret it.

frankboyu, 2025-04-15:
Sorry that I didn't read the code carefully enough. Deuteron is included in the functions ParticleMultiplexPower and DemultiplexPID. But it is listed in the decaying particles. I guess the real issue is deuteron being interpreted as decaying particle? Sorry about the confusion and I will keep looking at how to fix this.

aaust, 2025-04-17:
I agree with your assessment and don't see a reason against making the deuteron stable. Can you make a pull request for your proposal?

frankboyu, 2025-04-18:
Sure, but I still haven't found the place to edit this. As [inline static int ParticleMultiplexPower(Particle_t locPID)](https://github.com/JeffersonLab/halld_recon/blob/479927de15e9386b790cff178b05f82a78f3891b/src/libraries/include/particleType.h#L2781) function only gives an ID and the decaying flag seems to be set elsewhere? Do you happen to know where it is? Thanks again!

closed 2025-05-06 by frankboyu