JeffersonLab/halld_recon #935: jana:nskip causes exception

aaust, 2025-04-15:
I noticed that the new command line command to skip over events `jana:nskip` causes hd_root to thrown an exception:
```
10:47:13.973 [error] Exception on worker thread 0
10:47:14.449  [info] Processing paused.
10:47:14.449 [fatal] Exception in worker thread 0 while executing PhysicsEventMap2 on event #247830545
----------------------------------------------------------
JException
  Message:  GetSingle failed due to too many 22765845
  Function: JEventSource::GetObjects
  Class:    DEventProcessor_monitoring_hists
  Instance: DMCThrown
  Plugin:   monitoring_hists

Closed ROOT file
Exit code: 1
```

How to reproduce:
```
hd_root -Pjana:nskip=150000 -PPLUGINS=monitoring_hists /cache/halld/RunPeriod-2019-11/recon/ver01/REST/071881/dana_rest_071881_173.hddm
```

nathanwbrei, 2025-04-15:
I expect this is the same issue as JANA2/#403, which was originally addressed in JANA2/#409. Sadly that fix has its own problems, e.g. JANA2/#421, so I'm working on another round of fixes and testing.

aaust, 2025-07-14:
Fixed with [jana2.4.2](https://github.com/JeffersonLab/JANA2/releases/tag/v2.4.2), first used in [version_6.4.0.xml](https://halldweb.jlab.org/halld_versions/version_6.4.0.xml)

closed 2025-07-14 by aaust