JeffersonLab/halld_recon #938: Segfault in DChargedTrackHypothesis_factory::Correct_CDC_dEdx_amp

aaust, 2025-04-15:
In the currently running analysis launch, I see segmentation faults in DChargedTrackHypothesis_factory::Correct_CDC_dEdx_amp

The problem is repeatable, and occurs when using only monitoring hists. Examples for problematic files:
`/cache/halld/RunPeriod-2019-11/recon/ver01/REST/071881/dana_rest_071881_173.hddm` after ~180k events
`/cache/halld/RunPeriod-2019-11/recon/ver01/REST/071889/dana_rest_071889_077.hddm` after ~639k events
`/cache/halld/RunPeriod-2019-11/recon/ver01/REST/071922/dana_rest_071922_149.hddm` after ~779k events
`/cache/halld/RunPeriod-2019-11/recon/ver01/REST/071926/dana_rest_071926_267.hddm` after ~622k events

With version_6.1.1.xml, the issue can be reproduced with this command:
`hd_root -PPLUGINS=monitoring_hists -PNTHREADS=32 $FILE`

nsjarvis, 2025-04-15:
Hmm. I will take a look.  Thanks for the hints.

nsjarvis, 2025-04-16:
It crashes when it is fed with a nan.

The line calling the function is [here](https://github.com/JeffersonLab/halld_recon/blob/479927de15e9386b790cff178b05f82a78f3891b/src/libraries/PID/DChargedTrackHypothesis_factory.cc#L332)

I ran the first example, /cache/halld/RunPeriod-2019-11/recon/ver01/REST/071881/dana_rest_071881_173.hddm, single-threaded, with a print statement for the dedx and track angle theta added as the first line inside the function.  It crashes with theta_deg=nan (and dedx 4.6678e-06) sometime after 183228 events - I did not print the event number.

Now I don't know what to do with this.  I could make the function just exit if it is given nans, but it is not good to have a track in the system with an undefined angle.   I'll find the event number.

nsjarvis, 2025-04-16:
I suggest that we remove the nan-track-hypotheses as close to the source as possible, by ignoring them in the REST_reader.

nsjarvis, 2025-04-16:
I have a branch that implements the fix.  This is an excerpt from testing the first file mentioned previously, showing the event numbers generated by my eventnumber plugin, where it encounters the bad track hypothesis.
```
19:38:12.453  [info] 247901337
19:38:12.454  [info] 247901412
19:38:12.460  [info] 247901492
19:38:12.460  [info] suppressed nan
19:38:12.466  [info] 247901725
19:38:12.467  [info] 247901726
```
Would you like me to leave the 'suppressed nan' message in the code, write something more descriptive (eg 'ignoring track hypothesis containing nans') or remove it?

nsjarvis, 2025-04-17:
See [the PR](https://github.com/JeffersonLab/halld_recon/pull/945).   I chose "Ignored a track hypothesis containing nans".

closed 2025-04-17 by aaust