JeffersonLab/halld_recon #942: TDC-ADC timing monitoring plot with missing entries

zihlmann, 2025-04-16:
the lower panel of the monitoring plot has bins with missing entries seen as "gaps".
the horizontal axis labeling has been fixed but not yet loaded up to the gluon machine.
So the very last bin on the right that has entries in it is related to R95-S3 and R95-S4 is 
empty so are, R75-S10 to S16.

[image]

zihlmann, 2025-04-16:
R95-S3 is the TDC with the Start Counter paddles, R95-S4 is the TDC with the PSC counters

zihlmann, 2025-04-17:
the latest monitoring plot of run 131045 looks good

[image]

nsjarvis, 2025-04-17:
What is in R54-S3 ?

zihlmann, 2025-04-17:
<crate number="54" type="VXS">
    <slot number="1" type="CPU">
    </slot>
    <slot number="3" type="F1TDCV3">
      <channel number="0" detector="FDC_Wires" package="3" chamber="1" wire="25"/>
      <channel number="1" detector="FDC_Wires" package="3" chamber="1" wire="26"/>
      <channel number="2" detector="FDC_Wires" package="3" chamber="1" wire="27"/>

nsjarvis, 2025-04-17:
OK thanks.  I will take a look at the code.

nsjarvis, 2025-04-18:
I made this branch: [nsj_highlevel](https://github.com/JeffersonLab/halld_recon/blob/nsj_highlevel/src/plugins/monitoring/highlevel_online/JEventProcessor_highlevel_online.cc).   

Roc 33 Slot 3 is now back.  
Roc 51 slot 17 is empty because it only contains one TDC, RF in channel 34.  
Roc 95 slot 4 is empty because there isn't a template for the PSC.   I will leave that for @sdobbs 

[image]

sdobbs, 2025-04-18:
thanks @nsjarvis !  I can add the PSC - could you please make sure that all of your changes are pushed to GitHub?  right now I'm only seeing Alex's change

nsjarvis, 2025-04-18:
@sdobbs they are there now, sorry (forgot the commit!).  We should also remove 51-17.

nsjarvis, 2025-04-18:
Sean added in the PSC and removed the empty bin 51-17.  

For 5k events from run 131033 I now see this 

[image]

The pull request is [here](https://github.com/JeffersonLab/halld_recon/pull/948).