JeffersonLab/halld_recon #961: Unexpected crash with misspelled command line switch

sdobbs, 2025-04-29:
I accidentally ran the following command:

<pre>
[sdobbs@ifarm2401 src]$ hd_root -Pnthreads=20 -Pplugins=TAGM_TW,HLDetectorTiming,monitoring_hists /cache/halld/RunPeriod-2025-01/rawdata/Run131287/hd_rawdata_131287_001.evio -Pjana:nevents:20000
Invalid JANA parameter '-Pjana:nevents:20000': Expected format -Pkey=value
</pre>

this caught the error in the command line switch, but then as the command ran, it ended with a segfault:

<pre>
16:05:01.299 [fatal] Segfault detected!
16:05:01.299 [fatal] Hard exit due to segmentation fault! Backtrace:
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    0: std::thread::join()
16:05:01.299 [fatal]       /lib64/libstdc++.so.6:dbb38
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    1: JEventSource_EVIOpp::~JEventSource_EVIOpp()
16:05:01.299 [fatal]       hd_root:e1eb20
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    2: JEventSource_EVIOpp::~JEventSource_EVIOpp()
16:05:01.299 [fatal]       hd_root:e1f619
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    3: JComponentManager::~JComponentManager()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:22dca7
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    4: void __gnu_cxx::new_allocator<JComponentManager>::destroy<JComponentManager>(JComponentManager*)
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1ead89
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    5: void std::allocator_traits<std::allocator<JComponentManager> >::destroy<JComponentManager>(std::allocator<JComponentManager>&, JComponentManager*)
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1e9921
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    6: std::_Sp_counted_ptr_inplace<JComponentManager, std::allocator<JComponentManager>, (__gnu_cxx::_Lock_policy)2>::_M_dispose()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1e8c9f
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    7: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release()
16:05:01.299 [fatal]       hd_root:3d9f3a
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    8: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d7fc9
16:05:01.299 [fatal] 
16:05:01.299 [fatal]    9: std::__shared_ptr<JComponentManager, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d5722
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   10: std::shared_ptr<JComponentManager>::~shared_ptr()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d5764
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   11: JTopologyBuilder::~JTopologyBuilder()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:2185a8
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   12: void __gnu_cxx::new_allocator<JTopologyBuilder>::destroy<JTopologyBuilder>(JTopologyBuilder*)
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1eace1
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   13: void std::allocator_traits<std::allocator<JTopologyBuilder> >::destroy<JTopologyBuilder>(std::allocator<JTopologyBuilder>&, JTopologyBuilder*)
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1e9861
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   14: std::_Sp_counted_ptr_inplace<JTopologyBuilder, std::allocator<JTopologyBuilder>, (__gnu_cxx::_Lock_policy)2>::_M_dispose()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1e883d
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   15: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release()
16:05:01.299 [fatal]       hd_root:3d9f3a
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   16: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d7fc9
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   17: std::__shared_ptr<JTopologyBuilder, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d580c
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   18: std::shared_ptr<JTopologyBuilder>::~shared_ptr()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d5828
16:05:01.299 [fatal] 
[... 54 lines cut ...]
16:05:01.299 [fatal]   37: std::default_delete<JServiceLocator>::operator()(JServiceLocator*) const
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1dcca0
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   38: std::unique_ptr<JServiceLocator, std::default_delete<JServiceLocator> >::~unique_ptr()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d8dc4
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   39: JApplication::~JApplication()
16:05:01.299 [fatal]       /u/group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.5.0/jana/jana_2.4.0/Linux_Alma9-x86_64-gcc11.5.0/lib/libJANA.so:1d2524
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   40: main
16:05:01.299 [fatal]       hd_root:3b8715
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   41: ???
16:05:01.299 [fatal]       /lib64/libc.so.6:295d0
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   42: __libc_start_main
16:05:01.299 [fatal]       /lib64/libc.so.6:29680
16:05:01.299 [fatal] 
16:05:01.299 [fatal]   43: _start
16:05:01.299 [fatal]       hd_root:3c7285
16:05:01.299 [fatal] 
16:05:01.299 [fatal] 
</pre>

The command works fine if I specify the jana:nevents  switch properly or if I don't use it.

nathanwbrei, 2025-04-29:
Thanks for the heads-up!

sdobbs, 2025-04-30:
I think this might be a more general problem with parsing command line parameters.  I also just had a crash when I accidentally used the switch `-PTOF_online`

nathanwbrei, 2025-05-13:
Segfault is caused because JANA2 attempts to shut everything down (which includes call JEventSourceEVIOpp::Finish()) before having called JEventSourceEVIOpp::Open(). Finish() tries to do the correct thing and check whether the heap-allocated member variables are nullptr before accessing them. However, these member variables were never initialized, and random garbage evaluates to true. Fix is very easy.