JeffersonLab/halld_recon #969: CDC_online omits histograms when run on the CH gluons.

nsjarvis, 2025-05-02:
The CDC_online plugin makes only a few of its histograms when run on the gluons.  So far I have tried gluon150, 154, 155, all from the hdops account (because I don't have my a home directory @faustus123 if you have a minute please could you make one? Alex cannot). 

This is CDC_online’s output on gluon155
```
Attaching file test131345_001.root as _file0...
(TFile *) 0x1d5aab0
root [1] gDirectory->cd("CDC")
(bool) true
root [2] gDirectory->ls()
TDirectoryFile*		CDC	CDC
 KEY: TDirectoryFile	rings_occupancy;1	CDC rings: occupancy
 KEY: TH1I	cdc_num_events;1	CDC number of events
 KEY: TH2I	cdc_o;1	CDC occupancy by straw, ring
root [3] .q
```

and this is on ifarm
```
[njarvis@ifarm2402.jlab.org: ~/mywork/cdc_thr_scan ]> root test131345_001.root                                                                                                     root [0] 
Attaching file test131345_001.root as _file0...
(TFile *) 0x2c7ba50
root [1] gDirectory->cd("CDC")
(bool) true
root [2] gDirectory->ls()
TDirectoryFile*		CDC	CDC
 KEY: TDirectoryFile	rings_occupancy;1	CDC rings: occupancy
 KEY: TH1I	cdc_num_events;1	CDC number of events
 KEY: TH2I	cdc_o;1	CDC occupancy by straw, ring
 KEY: TH1I	cdc_raw_amp;1	CDC amplitude (ADC units, scaled)
 KEY: TH2I	cdc_raw_amp_vs_n;1	CDC amplitude (ADC units, scaled) vs straw number
 KEY: TH1I	cdc_raw_t;1	CDC raw time (units of 0.8ns)
 KEY: TH2I	cdc_raw_t_vs_n;1	CDC raw time (units of 0.8ns) vs straw number
 KEY: TH1I	cdc_raw_int;1	CDC integral (ADC units, scaled), pedestal subtracted
 KEY: TH2I	cdc_raw_int_vs_n;1	CDC integral (ADC units,scaled), pedestal subtracted, vs straw number
 KEY: TH1I	cdc_raw_intpp;1	CDC integral (ADC units, scaled), includes pedestal
 KEY: TH2I	cdc_raw_intpp_vs_n;1	CDC integral (ADC units, scaled), including pedestal, vs straw number
 KEY: TH1I	cdc_ped;1	CDC pedestal (ADC units)
 KEY: TH2I	cdc_ped_vs_n;1	CDC pedestal (ADC units) vs straw number
 KEY: TH1I	cdc_windata_ped;1	CDC pedestal (ADC units) from raw window data
 KEY: TH2I	cdc_windata_ped_vs_n;1	CDC pedestal (ADC units) from raw window data vs straw number
```

The behaviour on the gluons is the same whether I include other plugins or not.  The output for other more complicated plugins (eg CDC_Efficiency) was fine.

aaust, 2025-05-02:
I can confirm this issue on the ifarm when I compile halld_recon against root 6.32.08. Something must have changed in the behavior of root.
Is there a reason why the missing histograms are defined in BeginRun() and not in Init()?

nsjarvis, 2025-05-02:
I have no idea whether that was deliberate or accidental, it wa written, too long ago.  I cannot think of a reason to keep them there.

nsjarvis, 2025-05-07:
[In fact the reason is clear in the code.](https://github.com/JeffersonLab/halld_recon/blob/e99960e5234f06b0a0a57b401158430cd9868f03/src/plugins/monitoring/CDC_online/JEventProcessor_CDC_online.cc#L169)   In the early days of gluex running, we had to use modified f250 firmware for the f125.  When we started using the dedicated f125 firmware on 11 Sept 2015, the ranges of the readout quantities changed, and the easiest way to make the plugin compatible with both forms of the readout was to check the run number.  I don't think it's possible to get at anything in init.

This is not the only CDC code which has been kept compatible with both versions - the DAQ library still contains code for the original readout.  

Is it necessary to keep this plugin functional for the very old data?  I might compromise by commenting out the settings needed for the 2014 data.

Are there any other plugins with histograms defined in BeginRun?

sdobbs, 2025-05-07:
I think at this point, it's okay to have the code only work with data no earlier than 2016 ...

nsjarvis, 2025-07-28:
https://github.com/JeffersonLab/halld_recon/pull/1033

nsjarvis, 2025-07-30:
Fixed in https://github.com/JeffersonLab/halld_recon/pull/1033

closed 2025-07-30 by nsjarvis