JeffersonLab/halld_recon #989: cal_cal plugin not working within halld_recon

igjaegle, 2025-05-17:
The code is crashing because of the memory taking by the histogram booking. It should be noted when compiled as a private plugin, there is no issue.

aaust, 2025-05-19:
Most of the info is mentioned in PR #988

igjaegle, 2025-05-19:
You can find a jana cfg example here:
/w/halld-scshelf2101/home/gxproj4/calibration/cal/cal_cal/jana2-no-trk-data-ecal-gains-to-1.cfg

I am using this one for the actual calib of ECAL & FCAL.

tks ig.

sdobbs, 2025-05-19:
I pushed some small changes to a branch `ig-cal-cal-sdobbs`.  Please check to see if it works for you as well.

Basically, It looked like some arrays were defined to be too short.  I hope that this doesn't affect any current work.

Also, I noted the lines:

  TH1::SetDefaultSumw2();
  TH2::SetDefaultSumw2();

These do not seem to be like good practice, since these lines could affect other plugins besides this one.

igjaegle, 2025-05-19:
Is this already set somewhere else? If not how can one get the correct
error for the histogram filling the prompt and random background.

tks ig.

sdobbs, 2025-05-19:
Probably one has to set it for each histogram, individually.

igjaegle, 2025-05-19:
I wanted to avoid that but ok I will do that.

I tested the ig-cal-cal-sdobbs branch and it is still hanging and crashing.

tks ig.

sdobbs, 2025-05-21:
I was able to successfully run with the command `hd_root --loadconfigs /w/halld-scshelf2101/home/gxproj4/calibration/cal/cal_cal/jana2-no-trk-data-ecal-gains-to-1.cfg /cache/halld/RunPeriod-2025-01/rawdata/Run131523/hd_rawdata_131523_000.evio` 

Can you please give more information to try to reproduce this problem.

aaust, 2025-05-21:
I can confirm that Sean's branch is no longer killed for memory consumption. I also ran the above mentioned command on the ifarm successfully through the entire file.

igjaegle, 2025-05-21:
Hi, I pull the Sean's branch but was unable to compile the code due to this
error:

libraries/HDDM/DEventSourceREST.cc: In constructor
‘DEventSourceREST::DEventSourceREST(std::string, JApplication*)’:
libraries/HDDM/DEventSourceREST.cc:76:21: error: ‘class
JCalibrationGeneratorCCDB’ has no member named ‘SetApplication’
   76 |    calib_generator->SetApplication(app);
      |                     ^~~~~~~~~~~~~~

tks ig.

igjaegle, 2025-05-21:
Discard my previous message, I was using the wrong branch. I checkout
Sean's branch and pull the latest modif but still is hanging in the histo
booking part but maybe I am still doing something wrong.

tks ig.

aaust, 2025-09-19:
I believe, the latest changes in PR #1014 fixed this, right?