JeffersonLab/halld_recon #1018: MWPC Match Params vectors having only 1 entry

alfab3, 2025-07-01:
When calling 
`shared_ptr<const DFMWPCMatchParams>fmwpcparms=MuPhyp->Get_FMWPCMatchParams();`

Where
```
const DChargedTrackHypothesis *MuPhyp=tracks[ip]->Get_Hypothesis(MuonPlus);
if (MuPhyp==NULL) return;
const DTrackTimeBased *muplus=MuPhyp->Get_TrackTimeBased();
```
I try to call the variables in the detector match: 
`std::cout << "fmwpcparms->dLayers.size()= " << fmwpcparms->dLayers.size() << endl;`

and the output always says the size is 1 no matter what the event I look at, I am specifically testing fully reconstructed simulation as well as REST data. 

The output looks like this from the print statement:
fmwpcparms->dLayers.size()= 1

The effected custom plugin is here:
/w/halld-scshelf2101/home/alfab/cpp_analysis/jana2_plugins/UTILITIES/training_sample_gen_muon/JEventProcessor_training_sample_gen_muon.cc

staylorjlab, 2025-07-01:
When I get a chance I will look at what we can do in the REST output.   I need to refresh my memory about what I did for CPP runs.

alfab3, 2025-07-01:
I just did a quick test on a file that was not put through reconstruction (stopped after mcsmear) and I am getting the same error. 

Input file:
/volatile/halld/home/alfab/RunPeriod2022-05/simulation_files/bhmuons/bhmuons_w_accidental_smearing_1/hddm/smeared_w_accidentals/101582/geant4_smeared_101582_000.hddm

Output printed: 
fmwpcparms->dLayers.size()= 1

aaust, 2025-07-09:
Thank you for the detailed description. It seems, the code was misunderstood during the transition to jana2. We perform the clustering for each layer separately, but all clusters should be added to the list of FMWPCCluster objects. At the moment, the algorithm **sets** the FMWPCCluster objects for each layer, thus overwriting the previous one and only layer 6 remains.
https://github.com/JeffersonLab/halld_recon/blob/ae33f4ead9c4ea605eb128237ac05e2160dc4d2b/src/libraries/FMWPC/DFMWPCCluster_factory.cc#L194-L199
Instead, we should Insert(newCluster). I will confirm with the jana2 experts and provide a fix.

closed 2025-07-14 by aaust